archiver-appliance/epicsarchiverap #16: Exception transitioning archive pv state for pv test in state METAINFO_OBTAINED

MichaelRitzert, 2016-02-19:
I receive the following error in mgmt/logs/arch.log for all PVs:
2215033 [MgmtArchivePVWorkflow1] ERROR org.epics.archiverappliance.mgmt.archivepv.Archi
vePVState  - Exception transitioning archive pv state for pv test in state METAINFO_OBTAINED
java.lang.ClassCastException

Unfortunately, no context is provided in the message.

All PVs stay in "Initial sampling" and the Metrics report "Startup	In Progress".

This is a fresh instance set up using the single_machine_install.sh script.

slacmshankar, 2016-02-19:
For some reason, the samplingPeriod from the policies does not seem to be a float. Could you attach your policies file?

MichaelRitzert, 2016-02-19:
You're right, I just found the same. When editing the file, I changed the samplingPeriod to "1" for some PVs. Changing the policy to say "1.0" fixed the problem.

slacmshankar, 2016-02-19:
Excellent! Please let me know if you run into any issues.

closed 2016-02-19 by slacmshankar