archiver-appliance/epicsarchiverap #30: New BPL to specialised in getting PVstatus for all PVs

pozara, 2017-05-15:
Getting PV status for PVs is already possible using getPVstatus BPL, however when the number of PVs gets large, this becomes a very slow operation as it's not meant to be used in this manner.

Having a dedicated getAllPVsStatus (or simething like that) BPL which would be optimised to deal with large number of VPs would significantly reduce retrieval time.

slacmshankar, 2017-05-16:
I'm assuming you'd want a small number of fields for this call. For starters, we'll include

- Archival params creation time - this should indicate when the PV was added to the archiver and we  started archiving for the first time
 - Timestamp of last event from the IOC - this should be the last known timestamp for non-paused PV's
- Is this PV paused- Paused status of the PV.

Anything else?

pozara, 2017-05-16:
That sounds about right. And the pvName of course :)