archiver-appliance/epicsarchiverap #31: Add DESC field to pvdetails page

apanna, 2017-05-17:
It would be nice if one can see the detailed description of the pv (i.e. the PV's DESC field)  on the details page. 
Please refer to [tech-talks](http://www.aps.anl.gov/epics/tech-talk/2017/msg00812.php) for more details.

DanielCuneo, 2018-08-01:
I'd like to implement this feature. I'm looking at the source now and not sure where to start.
Please provide some guidance and I'll get started.

slacmshankar, 2018-08-01:
In ArchiveChannel.getPVMetrics, you may need to call getLatestMetadata and add that info for that channel into the PVMetrics. It may be worth it to refactor this a bit so that we can add various pieces of information to the PVMetrics object without having to create attributes for each piece of information. Please let me know if you need more info.