archiver-appliance/epicsarchiverap #33: Suspected memory leak

mbruker, 2017-07-11:
In a fresh installation of the most recent snapshot that was done using single_machine_install.sh, memory usage apparently increases with time at a rate of ~1 GB per day without any archiving activity. I assume this is unexpected behaviour but lack the knowledge of Java and Tomcat internals to diagnose it properly. The only message that looks suspicious to me is
> The web application [engine] appears to have started a thread named [JCA Command Thread] but has failed to stop it. This is very likely to create a memory leak.

Any help will be appreciated. I'll be happy to extract more files and diagnostic data but don't know where to start. Thanks in advance.

[catalina.err.txt](https://github.com/slacmshankar/epicsarchiverap/files/1138826/catalina.err.txt)
[arch.log.txt](https://github.com/slacmshankar/epicsarchiverap/files/1138834/arch.log.txt)

[image]

slacmshankar, 2017-07-11:
I see pretty much the same thing on some of my smaller instances. This really depends on the GC you are using; how aggressive you've configured your GC. If you are using the G1GC, you can control some of this using options like InitiatingHeapOccupancyPercent from http://www.oracle.com/technetwork/articles/java/vmoptions-jsp-140102.html but I generally leave these alone. Here's a snapshot from my test instance with a similar sawtooth pattern.
 
[image]

mbruker, 2017-07-13:
OK, thanks a lot for the clarification and sorry to have taken up your time. Even though the behaviour is not caused directly by your code, it might be a good idea to include a sentence or two about it in the documentation to avoid confusion :smiley:

closed 2017-08-24 by slacmshankar