archiver-appliance/epicsarchiverap #38: Postgres support

beenje, 2017-11-07:
Hello,
I work for ESS. We already use Postgres for several applications and would prefer to use it over MySQL/MariaDB for the Archiver Appliance as well. The documentation mentions "We hope to add Postgres support soon." Is that still planned?

I just saw that Redis support was added: https://github.com/slacmshankar/epicsarchiverap/releases/tag/v0.0.1_SNAPSHOT_22-June-2016
As I understand, that would be an alternative to MariaDB. Correct?
Is that considered ready to use?

Thanks.

slacmshankar, 2017-11-07:
>> We hope to add Postgres support soon."

Another lab was considering adding support for Postgres; however, they decided to use CSS instead..

>> As I understand, that would be an alternative to MariaDB

Yes; this would be instead of MariaDB. There is support for other config stores as well.

I estimate that it should be reasonably trivial to add support for Postgres. You need an implementation of the ConfigPersistence interface - http://slacmshankar.github.io/epicsarchiver_docs/api/org/epics/archiverappliance/config/ConfigPersistence.html

It should be easy to use MySQLPersistence as the basis for the code and use the Postgres text  datatype - https://www.postgresql.org/docs/9.1/static/datatype-character.html?

Please let me know if you'd be interested. The hardest part for this is integration testing, and if you are willing to integrate test this; this should be simple to add...

Regards,

Murali

beenje, 2017-11-09:
Thanks for your quick reply.
After some internal discussion, we unfortunately don't have the resources to work on Postgres support right now.

I have some experience with Redis. I think it could be a nice alternative.
I'll probably try to test it when I have some time.

Thanks

Benjamin

jacomago, 2024-03-18:
Unneeded feature now.

closed 2024-03-18 by jacomago