archiver-appliance/epicsarchiverap #39: Different time between getData and ArchiveViewer

gabrielfedel, 2017-11-28:
I'm not sure if it is a bug, and if it is from epicsarchiver or archiver viewer. 

When a get data from a specific PV using archiver viewer, and from API getData I have diffferent timestamps (val is different too, on getData result I have few precision).

I'm using Oct 12, 2016 version.

slacmshankar, 2017-11-28:
When you say viewer, which viewer would be referring to? Also, are you getting the data from the server as JSON? An example and some hints on reproducibility would help.

gabrielfedel, 2017-11-29:
I'm referred to Archiver Viewer (the defaul viewer, https://screenshots.firefox.com/BtKpxVjrG7uplBHN/10.2.105.191).
I'm getting data from server as JSON, and get data from Archiver Viewer as csv.

I could find this "bug" with one specific PV, I'm not sure which information to pass to reproduce this error.

gabrielfedel, 2017-11-29:
To complement, here a piece of data from getData.json:

1511909530 | 1.6E-9
1511909534 | 1.5E-9
1511913022 | 1.6E-9
1511913024 | 1.5E-9
1511954098 | 1.6E-9
1511954100 | 1.5E-9
1511960384 | 1.6E-9
1511960386 | 1.5E-9
1511909314 | 1.6E-9
1511909354 | 1.5E-9
1511909358 | 1.6E-9
1511909366 | 1.5E-9
1511909368 | 1.6E-9
1511909370 | 1.5E-9

And here a piece of data from Archiver Viewer:

XAFS2:MKS937A1:P1 | 1511716940 | 5000000000 | 1.45e-9
XAFS2:MKS937A1:P1 | 1511747737 | 5000000000 | 1.45e-9
XAFS2:MKS937A1:P1 | 1511778534 | 5000000000 | 1.5e-9
XAFS2:MKS937A1:P1 | 1511809331 | 5000000000 | 1.55e-9
XAFS2:MKS937A1:P1 | 1511840128 | 5000000000 | 1.55e-9
XAFS2:MKS937A1:P1 | 1511870925 | 5000000000 | 1.55e-9
XAFS2:MKS937A1:P1 | 1511901722 | 5000000000 | 1.55e-9
XAFS2:MKS937A1:P1 | 1511932519 | 5000000000 | 1.55e-9
XAFS2:MKS937A1:P1 | 1511963316 | 5000000000 | 1.55e-9

gabrielfedel, 2017-11-30:
The Archiver Viewer repeat the last value acquired, when the PV is disconnected ?