archiver-appliance/epicsarchiverap #41: Bug in sampleStartup.sh of archiverAppliance

slacmshankar, 2017-12-07:
Hello,

I have migrated an archiver appliance from one machine/filesystem to another 
machine/filesystem. Everything works fine except I can not retrieve values 
from within CSS. Must be a missconfiguration.
Searching for the reason, I found a bug in the sample startup script.

In the startup function:

function startTomcatAtLocation() { 
    if [ -z "$1" ]; then echo "startTomcatAtLocation called without any 
arguments"; exit 1; fi
    export CATALINA_HOME=$TOMCAT_HOME
    export CATALINA_BASE=$1
    echo "Starting tomcat at location ${CATALINA_BASE}"
    
    ARCH=`uname -m`
    if [[ $ARCH == 'x86_64' || $ARCH == 'amd64' ]]
    then
      echo "Using 64 bit versions of libraries"
      export LD_LIBRARY_PATH=${CATALINA_BASE}/webapps/engine/WEB-INF/lib/
native/linux-x86_64:${LD_LIBRARY_PATH}
    else
      echo "Using 32 bit versions of libraries"
      export LD_LIBRARY_PATH=${CATALINA_BASE}/webapps/engine/WEB-INF/lib/
native/linux-x86:${LD_LIBRARY_PATH}
    fi
    
    pushd ${CATALINA_BASE}/logs
    ${CATALINA_HOME}/bin/jsvc \
        -server \
        -cp ${CATALINA_HOME}/bin/bootstrap.jar:${CATALINA_HOME}/bin/tomcat-
juli.jar \
        ${JAVA_OPTS} \
        -Dcatalina.base=${CATALINA_BASE} \
        -Dcatalina.home=${CATALINA_HOME} \
        -cwd ${CATALINA_BASE}/logs \
        -outfile ${CATALINA_BASE}/logs/catalina.out \
        -errfile ${CATALINA_BASE}/logs/catalina.err \
        -pidfile ${CATALINA_BASE}/pid \
        org.apache.catalina.startup.Bootstrap start
     popd
}

Here LD_LIBRARY_PATH gets set to an invalid path, as CATALINA_BASE is set from 
the parameter passed to the function:

which is called like:

startTomcatAtLocation ${ARCHAPPL_DEPLOY_DIR}/mgmt

So in the end the path looks like:

${ARCHAPPL_DEPLOY_DIR}/mgmt/webapps/engine/WEB-INF/lib/native/linux-x86...

But the real PATH here should be 

${ARCHAPPL_DEPLOY_DIR}/mgmt/webapps/mgmt/WEB-INF/lib/native/linux-x86...

The archiver still works without the LD_LIBRARY_PATH set correctly.

Regards,

Jörn

frazor, 2018-08-27:
Hi! Any thoughts on my uploaded patch?

slacmshankar, 2018-08-27:
Sorry; missed this. Will respond by the end of the week.

PR #58 (2022-11-21): Adjust LD_LIBRARY_PATH to the right service
In sampleStartup.sh LD_LIBRARY_PATH gets set to an invalid path, as it
contains path to engine instead of mgmt.

**Warning**: I haven't run any test cases on this yet, I don't know ant nor the test cases very well. Please let me know in case I can assist by running some test command on this code to get it accepted.

> So in the end the path looks like:
> 
> ${ARCHAPPL_DEPLOY_DIR}/mgmt/webapps/**engine**/WEB-INF/lib/native/linux-x86...
> 
> But the real PATH here should be
> 
> ${ARCHAPPL_DEPLOY_DIR}/mgmt/webapps/**mgmt**/WEB-INF/lib/native/linux-x86. [... cut]
commit 1aa26a76 (2022-11-21): Adjust LD_LIBRARY_PATH to the right service

In sampleStartup.sh LD_LIBRARY_PATH gets set to an invalid path, as it
contains path to engine instead of mgmt.

So in the end the path looks like:

${ARCHAPPL_DEPLOY_DIR}/mgmt/webapps/engine/WEB-INF/lib/native/linux-x86...

But the real PATH here should be

${ARCHAPPL_DEPLOY_DIR}/mgmt/webapps/mgmt/WEB-INF/lib/native/linux-x86...

The archiver still works without the LD_LIBRARY_PATH set correctly.

Fixes: #41
closed 2022-11-21 by slacmshankar