archiver-appliance/epicsarchiverap #47: Waveform limitation

yannickC, 2018-01-29:
Hello,

I'm trying to add a waveform (with DTYP: asynInt16ArrayIn) in the Archive Appliance but I can't, because the size seems to big.

The size of my waveform is 2304000 elements. With a smaller one, it works.

Is it possible to modify this limitation ?

Regards
Yannick

slacmshankar, 2018-01-29:
Possibly. Do you see anything in the logs?

yannickC, 2018-01-30:
Hello,

I have this in /engine/logs/arch.log :

1481901 30 janv. 2018 11:51:20, 656 [JCA Command Thread] ERROR org.epics.archiverappliance.engine.pv.EPICS_V3_PV  - exception when subscribing pv NPM:VCAMOBSIMAGE-ArrayData
java.lang.IllegalArgumentException: Size of the monitor exceeds maxArrayBytes (2304000 * 2 > 16384)
    at gov.aps.jca.Channel.checkMonitorSize(Channel.java:681)
    at com.cosylab.epics.caj.CAJChannel.addMonitor(CAJChannel.java:1064)
    at org.epics.archiverappliance.engine.pv.EPICS_V3_PV.subscribe(EPICS_V3_PV.java:453)
    at org.epics.archiverappliance.engine.pv.EPICS_V3_PV.access$700(EPICS_V3_PV.java:55)
    at org.epics.archiverappliance.engine.pv.EPICS_V3_PV$1$1.run(EPICS_V3_PV.java:256)
    at org.epics.archiverappliance.engine.pv.JCACommandThread.run(JCACommandThread.java:205)
1541801 30 janv. 2018 11:52:20, 556 [Scheduler for the whole engine] ERROR org.epics.archiverappliance.engine.metadata.MetaGet  - Cannot determine DBR type for pv NPM:VCAMOBSIMAGE-ArrayData

Where I can change this limitation ?

Regards,
Yannick

slacmshankar, 2018-01-30:
I think this is set using EPICS_CA_MAX_ARRAY_BYTES in the EPICS configuration. We also print this on engine startup. So, you should be able to change this by changing the EPICS_CA_MAX_ARRAY_BYTES environment variable? I think that this change probably needs to be made on both the archiver and IOC side for things to work correctly...

<context class="com.cosylab.epics.caj.CAJContext">
  <preemptive_callback>true</preemptive_callback>
.......
  <max_array_bytes>80000000</max_array_bytes>
  <event_dispatcher class="org.epics.archiverappliance.engine.epics.JCAEventDispatcherBasedOnPVName"/>
</context>

slacmshankar, 2018-04-09:
Probably not an issue with the archiver.. Please let me know if otherwise.

closed 2018-04-09 by slacmshankar