archiver-appliance/epicsarchiverap #49: getPVStatus ignores POST body

willrogers, 2018-03-05:
The documentation for the `getPVStatus` BPL call suggests that you can send the requested PVs in a POST. As far as I can tell, it is ignoring the post and it works only if the PVs are added to the URL.

I can figure out how to fix this if you like @slacmshankar.

slacmshankar, 2018-03-06:
You are right; I think this is a bug. 

I'll try to fix it by the end of this week. Please feel free to fix/ping me if I do not get back to you.

slacmshankar, 2018-03-06:
This does seem to support application/x-www-form-urlencoded POST's. However, it does not support the full fledged POST with JSON etc. Is that what you were looking for?

```
$ cat pvs.txt
pv=XCOR:LI21:101:BDES,XCOR:LI21:101:EDES,XCOR:LI21:402:BDES
$ curl -XPOST --data @pvs.txt "http://localhost:17665/mgmt/bpl/getPVStatus"
[
{"lastRotateLogs":"Never","appliance":"test-arch","pvName":"XCOR:LI21:101:BDES","pvNameOnly":"XCOR:LI21:101:BDES","connectionState":"true","lastEvent":"Mar\/05\/2018 17:13:58 -08:00","samplingPeriod":"1.0","isMonitored":"true","connectionLastRestablished":"Never","connectionFirstEstablished":"Mar\/05\/2018 17:08:00 -08:00","connectionLossRegainCount":"0","status":"Being archived"},
{"lastRotateLogs":"Never","appliance":"test-arch","pvName":"XCOR:LI21:101:EDES","pvNameOnly":"XCOR:LI21:101:EDES","connectionState":"true","lastEvent":"Mar\/05\/2018 17:08:00 -08:00","samplingPeriod":"1.0","isMonitored":"true","connectionLastRestablished":"Never","connectionFirstEstablished":"Mar\/05\/2018 17:08:00 -08:00","connectionLossRegainCount":"0","status":"Being archived"},
{"lastRotateLogs":"Never","appliance":"test-arch","pvName":"XCOR:LI21:402:BDES","pvNameOnly":"XCOR:LI21:402:BDES","connectionState":"true","lastEvent":"Mar\/05\/2018 17:08:00 -08:00","samplingPeriod":"1.0","isMonitored":"true","connectionLastRestablished":"Never","connectionFirstEstablished":"Mar\/05\/2018 17:08:00 -08:00","connectionLossRegainCount":"0","status":"Being archived"}]
$ 
```

closed 2018-03-06 by willrogers