archiver-appliance/epicsarchiverap #52: AA chart shows empty graph if the PV hasn't changed.

abdalla-dalleh, 2018-04-29:
It is known that AA does not add new sample to PB file if the PV hasn't changed. The problem is, this will result in an empty chart if you requested an interval where the PV hasn't changed since the last archived value. This is done very well in CSS data browser where it will show a straight line indicating that the PV hasn't changed in this interval. I really would like you to consider drawing the chart and keeping it continuous even if the PV hasn't changed.

frazor, 2018-06-11:
I too have a user at ESS seeing the same for very constant temperature data. Device reports, but always the same value. And the data line goes away when the first data point is no longer in the zoom.
We too would like to have a line and possibility to see the other data points/have a straight chart.

willrogers, 2018-08-23:
I agree with this proposal.

willrogers, 2018-08-28:
I guess you're right. This looks harmless to me as it is a semi-genuine data-point. The assumption is that the value is still the same as the last one in the archive. If you pan to the right the trace should end at that point i.e. not plot into the future.

slacmshankar, 2018-08-28:
Removed the add(1, "days") from the previous diff and checked in. Confirmed trace does end at the end time when we pan to the right.

closed 2018-08-28 by slacmshankar