archiver-appliance/epicsarchiverap #54: Lots of Hazelcast log warnings

willrogers, 2018-05-21:
I'm getting a lot of warning messages in `mgmt.log` from Hazelcast:

``` WARN com.hazelcast.client.impl.ClientHeartbeatMonitor - Client heartbeat is timed out, closing connection to Connection[id=6, 0.0.0.0/0.0.0.0:17660->null, endpoint=[172.23.194.62]:53162, alive=false, type=JAVA_CLIENT]. Now: 2018-05-18 13:33:31.660. LastTimePacketReceived: 2018-05-18 13:33:31.660
```

I'm not certain, but I think these are erroneous. This Hazelcast ticket suggests it's fixed in newer versions : https://github.com/hazelcast/hazelcast/issues/8777.

Could we upgrade to a newer version of Hazelcast?

slacmshankar, 2018-05-21:
Thanks for the heads up. Will update this by the end of the week.

closed 2018-06-14 by slacmshankar

willrogers, 2018-09-05:
Just a note to say that this appears to have resolved the problem, thanks.