archiver-appliance/epicsarchiverap #55: PVs stuck at "appliance assigned"

djvine, 2018-05-23:
Hi,

I used the single machine install script and started the archiver using the sample startup script and there was no problems. I added some PVs but they never move from "appliance assigned" to "being archived".

The "May not exist" report does not currently show the PVs (although it did prior to restarting the appliance). "Recently modified PVs" shows no entries.

I guess it is unable to connect to the PVs. Running caget from the terminal connects no problem. 

Can anyone recommend how to proceed to troubleshoot this problem?

Thanks,
David

slacmshankar, 2018-05-23:
Most likely your EPICS configuration needs to be fixed. The engine's logs will print out the details of the EPICS configuration like so...
<context class="com.cosylab.epics.caj.CAJContext">
  <preemptive_callback>true</preemptive_callback>
  <addr_list>129.1.1.255</addr_list>
  <auto_addr_list>false</auto_addr_list>
  <connection_timeout>30.0</connection_timeout>
  <beacon_period>15.0</beacon_period>
  <repeater_port>5068</repeater_port>
  <server_port>5078</server_port>
  <max_array_bytes>80000000</max_array_bytes>
  <event_dispatcher class="org.epics.archiverappliance.engine.epics.JCAEventDispatcherBasedOnPVName"/>
</context>
You can check to see if this agrees with you configuration. All of these should come from the standard EPICS variables as defined here - https://epics.anl.gov/base/R3-14/12-docs/CAref.html#EPICS

djvine, 2018-05-23:
Thank you for your fast reply!

I found the solution here: https://epics.anl.gov/tech-talk/2017/msg00695.php

I didn't realize the sampleStartup.sh needed to be run as root. Using sudo fixed the problem.

Thanks!

slacmshankar, 2018-05-23:
Ok; please let me know if this is still an issue.

closed 2018-05-23 by slacmshankar

djvine, 2018-05-23:
No, you can close it thanks.