archiver-appliance/epicsarchiverap #57: Unit Test not building

toomanycats, 2018-06-07:
I've downloaded the stable release, built the WAR files with ant.
Loaded the eclipse .project file into Eclipse.

No obvious errors showing in Eclipse.
Tomcat 7 installed and running on correct port.
Copied WAR files to Tomcat installation webapps directory.

All unit tests are failing to build. Must be some configuration that I have not done, however doc mentions nothing so it's very difficult to trouble shoot as a new Java user.

In a terminal I type:
    ant -Dtest=SimpleETLTest  singletest      
Buildfile: ~/epicsarchiverap-0.0.1_SNAPSHOT_27-Nov-2017/build.xml
     [echo] Building the archiver appliance for the site tests

compile:
    [javac] Compiling 1 source file to ~/epicsarchiverap-0.0.1_SNAPSHOT_27-Nov-2017/bin

singletest:
    [junit] Running org.epics.archiverappliance.engine.test.ChannelTest
    [junit] Testsuite: org.epics.archiverappliance.engine.test.ChannelTest
    [junit] 0    [main] INFO  org.epics.archiverappliance.SIOCSetup  - Starting SIOC with DB file ~/epicsarchiverap-0.0.1_SNAPSHOT_27-Nov-2017/./src/test/org/epics/archiverappliance/UnitTestPVs.db
    [junit] 135  [main] INFO  config.org.epics.archiverappliance.config.ConfigServiceForTests  - The WEB-INF/classes folder is /home/dpcuneo/epicsarchiverap-0.0.1_SNAPSHOT_27-Nov-2017/./bin
    [junit] null
    [junit] Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 0.142 sec
    [junit] Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 0.142 sec
    [junit] ------------- Standard Output ---------------
    [junit] 0    [main] INFO  org.epics.archiverappliance.SIOCSetup  - Starting SIOC with DB file ~/epicsarchiverap-0.0.1_SNAPSHOT_27-Nov-2017/./src/test/org/epics/archiverappliance/UnitTestPVs.db
    [junit] 135  [main] INFO  config.org.epics.archiverappliance.config.ConfigServiceForTests  - The WEB-INF/classes folder is /home/dpcuneo/epicsarchiverap-0.0.1_SNAPSHOT_27-Nov-2017/./bin
    [junit] null
    [junit] ------------- ---------------- ---------------
    [junit] 
    [junit] Testcase: testAll took 0.139 sec
    [junit]     Caused an ERROR
    [junit] Exception loading installation specific properties file archappl.properties from classpath
    [junit] org.epics.archiverappliance.config.exception.ConfigException: Exception loading installation specific properties file archappl.properties from classpath
    [junit]     at org.epics.archiverappliance.config.ConfigServiceForTests.<init>(ConfigServiceForTests.java:96)
    [junit]     at org.epics.archiverappliance.config.ConfigServiceForTests.<init>(ConfigServiceForTests.java:62)
    [junit]     at org.epics.archiverappliance.engine.test.ChannelTest.setUp(ChannelTest.java:44)
    [junit] Caused by: java.lang.NullPointerException
    [junit]     at java.util.Properties$LineReader.readLine(Properties.java:434)
    [junit]     at java.util.Properties.load0(Properties.java:353)
    [junit]     at java.util.Properties.load(Properties.java:341)
    [junit]     at org.epics.archiverappliance.config.ConfigServiceForTests.<init>(ConfigServiceForTests.java:93)
    [junit] 

BUILD FAILED
~/epicsarchiverap-0.0.1_SNAPSHOT_27-Nov-2017/build.xml:272: Test org.epics.archiverappliance.engine.test.ChannelTest failed

Total time: 1 second

I don't understand why the WEB-INF directory passed to the config object is "./bin".
Where should I be running this test ?

frazor, 2018-06-11:
Hi @toomanycats;
I'm a layman, but I'm unable to repeat your error unfortunately.

I have CentOS 7.4 with ant and ant-junit installed on the system. I have configured both CATALINA_HOME and TOMCAT_HOME to /usr/share/tomcat which is the tomcat basedir under CentOS.

I get the following instead, it seems your system launches the wrong tests somehow?:

[vagrant@localhost epicsarchiverap]$ git log -1 --decorate --oneline
98ab374 (HEAD, tag: **v0.0.1_SNAPSHOT_27-Nov-2017**) Make the workflow transitions s
[vagrant@localhost epicsarchiverap]$ **java -version**
openjdk version "1.8.0_161"
OpenJDK Runtime Environment (build 1.8.0_161-b14)
OpenJDK 64-Bit Server VM (build 25.161-b14, mixed mode)
[vagrant@localhost epicsarchiverap]$ **ant -Dtest=SimpleETLTest singletest**
Buildfile: /myess/epicsarchiverap/build.xml
     [echo] Building the archiver appliance for the site tests

compile:

singletest:
    [junit] Running **org.epics.archiverappliance.etl.SimpleETLTest**
    [junit] Testsuite: **org.epics.archiverappliance.etl.SimpleETLTest**
    [junit] 0    [main] INFO  config.org.epics.archiverappliance.config.ConfigServiceForTests  - The WEB-INF/classes folder is /myess/epicsarchiverap/./bin
    _[junit] 13   [main] INFO  config.org.epics.archiverappliance.config.ConfigServiceForTests  - Done loading installation specific properties file file:/myess/epicsarchiverap/bin/archappl.properties_
    [junit] 13   [main] WARN  org.epics.archiverappliance.config.ConfigServiceForTests  - Overriding JCA command thread count to 1
    [junit] 16   [main] INFO  config.org.epics.archiverappliance.config.ConvertPVNameToKey  - The pv name components in this installation are separated by these characters [\:\-] and the key names are terminated by :

I really just copied your command line, so not sure what's wrong. Perhaps you can throw us some software versions, java, ant, etc?

frazor, 2018-06-11:
... and much later my run ends well:

```
...
    [junit] Testcase: testMove took 2,634.282 sec

BUILD SUCCESSFUL
Total time: 43 minutes 59 seconds
[vagrant@localhost epicsarchiverap]$ 

```
I hope you too can make it work on your machine.

slacmshankar, 2018-06-12:
For the unit tests, we use getResourceAsStream to pick up the archappl.properties. Or, in other words, we are using the class loader infrastructure to load this file. When you compile, all the war file build steps in the build.xml will copy the archappl.properties into the ./bin folder and we should load it from there. To run the unit tests, you will need to compile. I typically do

ant; ant -Dtest=SimpleETLTest singletest.

Hope that helps.

frazor, 2018-06-12:
I forgot to add that I built my project with just `ant` first as well. Doh. :-)

toomanycats, 2018-06-12:
@slacmshankar 
Thanks for responding.

> When you compile, all the war file build steps in the build.xml will copy the archappl.properties into the ./bin folder and we should load it from there.

That's my problem. I do not see anything in `/bin` .  My build command must have been incorrect or incomplete.

**UPDATE**:
I rebuilt with `ant` and the previous test which failed, has now succeeded.

Suggested Changes
- If the config path returned is NULL
-- Instruct the user to check on the contents of `./bin` and rebuild.
- When running unit test
-- Print the path where the test logs will be written.

I'll  issue a pull request ASAP with such minor changes. I've not done that before, perhaps this is a good starting place.

Cheers

slacmshankar, 2018-06-12:
These are good suggestions; look forward to the PR.

closed 2018-06-12 by slacmshankar