archiver-appliance/epicsarchiverap #70: Archived data because the sampling buffer capacity is too small and lost a lot, only part of the data is normal.

alonelytree, 2018-11-30:
@slacmshankar  hi,shankar ,I have a question to ask you，A problem was discovered today during the operation of the new archive system，The data curve circulates like a heartbeat in a specific cycle, and the time axis is enlarged. It can be clearly seen that within ten seconds, there are only eleven sampling points, and the other time data is a straight line，all the data is in this case,View archive pv details，the sample buffer capacity is 11,many events lost because the sample buffer is full.Can there be any way to increase the sampling buffer capacity and prevent data loss?Sorry, the system can't upload photos, or there will be pictures that will clearly describe the problem.

I look forward to your reply,。
Thank you very much.

slacmshankar, 2018-11-30:
You should be able to control the size of the buffers by changing the PV's sampling period. Please see question 4 in the FAQ - http://slacmshankar.github.io/epicsarchiver_docs/faq.html. There is also a global setting in archappl.properties that applies to app PVs.

alonelytree, 2018-12-10:
Ok,Thank you.

closed 2018-12-10 by alonelytree