archiver-appliance/epicsarchiverap #73: Inserting non-epics variables into the Archive Appliance database.

CarlosCumming, 2019-01-03:
The old (very old) Channel Archiver had an interface (casi) that would allow insertion of non-epics data given a "pvish" name, value and timestamp. We still use casi, and this issue is holding us back from adapting the Archiver Appliance.

slacmshankar, 2019-01-07:
Are you looking to implement an interface that would let the appliance "connect" to these PVs? Or some examples on how to write .pb files.

CarlosCumming, 2019-01-08:
The casi interface in the old Channel Archiver allows you to write into the Channel Archiver database. The values written then appear as if they were archived process variables.

If there already is a high level interface that allows writing to the .pb files, I guess that would fulfill our requirements - assuming this interface takes care of file locking, and any other nasty-isms. I already looked for .pb file interface documentation and didn't find any. Maybe I missed it?

On the other hand, I did consider making a sioc to act as a, "bridge" between epics and non-epics data.  I'm still considering this as a possible alternative. I think that's what your alluding to <above>.

thx, tom.c

slacmshankar, 2019-01-08:
It should be theoritically possible to add a PV to the archiver and then write the data to a .pb file etc. I can send you samples but it will still be some work. But like you suggest; it might be faster/simpler to use pcaspy or p4p to create a bridge between non-EPICS PV's and the archiver.

CarlosCumming, 2019-01-08:
The more I dig into this, the pcaspy option looks like a viable option.

I think this case could be closed, unless I find a major issue with it and I can open a new case.

closed 2019-01-08 by CarlosCumming