archiver-appliance/epicsarchiverap #82: Support for Java 12/Tomcat 9 etc

slacmshankar, 2019-05-21:
Upgrade the system requirements to more recent versions of the JDK/Tomcat/other libs. This may require folks to upgrade their JDK/Tomcat and may not be backwards compatible (in terms of system requirements).

slacmshankar, 2019-05-22:
Migrated the source to be compatible with Java 12. Needed some work on the JavaDoc as the Taglet API has changed significantly. Tested with Tomcat 7.0.94; Tomcat 7 seems to come up with Java 12.

slacmshankar, 2019-05-22:
Am able to bring up the archiver on Tomcat 9 as well. Needed to change the connection pool settings to cater to DBCP2 but very small number of changes. Tested using mysql-connector-java-5.1.47.jar.

pklaus, 2020-06-29:
I guess this can be closed as the **JDK 12/Tomcat 9 release** was [published on Nov 14 2019](https://github.com/slacmshankar/epicsarchiverap/releases/tag/v0.0.1_SNAPSHOT_13-Nov-2019).

beenje, 2020-10-29:
Hello! Is there any specific reason Java 12 was targeted? And not 11?
The current LTS version is 11 . Other versions only have 6 months availability. Java 12 end of availability was Sept 2019.
See https://adoptopenjdk.net/support.html

The latest archiver appliance specifies jdk 12+. Is it really incompatible with jdk 11 (LTS)?

slacmshankar, 2022-10-05:
We've moved on to JDK 16. I think JDK 12 was targeted mostly because it was the latest at that time. The main transition was beyond JDK 9+ as that's when all of the module changes happened and also Tomcat 9+ with its support system.

closed 2022-10-05 by slacmshankar