archiver-appliance/epicsarchiverap #83: define access rights for storage

dschick, 2019-06-04:
Hej,

I am running the archiver as a service in my debian, so essentially it is run as root.
This leads to the fact that all data is stored in the data locations by root as well with no access right to other users.
`-rw------- 1 root root 4235 Jun  4 11:26 RP:2019_06_04_09.pb`
I would like to change that to allow reading for everyone.

Is this possible?

slacmshankar, 2019-06-04:
If you are checking to see if there is some hook in the archiver to change permissions, sorry no..

However, why do you run as root? All service startup scripts let you define the user you run the process as. Also, you can set the proper umask in your startup script to allow read's by all..

dschick, 2019-06-05:
okay, that was simple. Thanks a lot for the quick response!

dschick, 2019-06-05:
btw, the User and Group are working for me, the UMask somehow not. Maybe the UMask is somewhere overwritten?

slacmshankar, 2019-06-05:
I don't think we do anything with the umask; however, Tomcat does have a setting in it's startup file. My version of Tomcat has this environment variable that can be used
#   UMASK           (Optional) Override Tomcat's default UMASK of 0027
#

slacmshankar, 2019-06-13:
Did this work?

dschick, 2019-06-13:
Sorry, I did not have time to try it yet.

closed 2019-07-24 by slacmshankar