archiver-appliance/epicsarchiverap #88: How to access historical PV data if the PVs are removed from the system

jokinea, 2019-09-30:
Hi there, I've got funny problem. I have archived PV "x" for two years and I have got a nice database with the PV "x" related data. 

Now if I remove the PV "x" from my system how I can still access the historical data? Of course as the work-around I could just keep the PV "x" running as a ghost in the background but is there any other way around this? Thanks a lot!

slacmshankar, 2019-09-30:
Will the retiredPVTemplate in getData call (see http://slacmshankar.github.io/epicsarchiver_docs/userguide.html) be of use?

jokinea, 2019-10-08:
Hi, ahaha sorry. It seems that my reading ability was the problem in the end. This seems to be exactly what we needed.

closed 2019-10-08 by jokinea