archiver-appliance/epicsarchiverap #89: Cannot find viewer.zip after a build

nicdevel, 2019-10-25:
Hi there,
I'm new to github.
I just tried to build the archiver using the source code in git. I used ant with Oracle Java(JDK) 12 and Tomcat 9. The retrieval.war file is generated with a warning:

[zip] Warning: skipping zip archive /opt/epicsarchiverap/stage/org/epics/archiverappliance/retrieval/staticcontent/viewer.zip because no files were included.

Due to this, I'm getting error 404 from the retrieval link, as this file is not present.
I checked if the files needed existed (looked inside an existing SNAPSHOT), these files are not present into the source code, too. Is it ok?
Did I do something wrong?

closed 2019-10-25 by nicdevel

jeonghanlee, 2019-10-25:
You may try the following commands:
```
git clone https://github.com/slacmshankar/epicsarchiverap
git submodule update --init --reference ./
```

slacmshankar, 2019-10-25:
Can you try
git clone --recursive https://github.com/slacmshankar/epicsarchiverap
This gets the viewer submodule

nicdevel, 2019-10-28:
Done it, thanks!