archiver-appliance/epicsarchiverap #92: PV can't automatically restore the connection

HongYi-Shern, 2019-11-12:
Hi, 
In some cases, when the machine is down or the device is faulty, the archive system will be disconnected from the ca server. When the fault is recovered, the pv cannot automatically restore the connection. what can be done to solve this problem?

Thank you.

slacmshankar, 2019-11-12:
The archiver should automatically reconnect back to the PV. It may however take some time if you have a lot of disconnected PV's. CA throttles the searches down if the IOC has been inactive for a while. When everything is working, when the IOC starts up, the CA client in the archiver should detect a beacon anomaly and kick off searches for all disconnected PV's. I would check to see if the repeater is configured correctly etc.

If you want to force a reconnect and you have a list of PV's, you can simply pause and resume the PV. This forcibly shutdown's the CA channel and recreates it and usually results in an immediate reconnection.

Hope that helps.

HongYi-Shern, 2019-11-13:
However, in the process of using the archiver appliance, when the connection is disconnected due to downtime or equipment failure, power on or failure recovery, pv will not automatically reconnect.

slacmshankar, 2019-11-13:
In the PV details page for the PV, there should be a line item that looks like so:
_PVs Internal State	Connecting **Searches: 52**_
The number of searches is the number of CA searches that were sent out; this should increase slowly with time (a few searches per hour) if the archiver is sending out search requests. Is this happening?

HongYi-Shern, 2019-11-22:
I observe for a while ,some pv can't automatically restore the connection ,but pause and resume it ,the pv will reconnect.

HongYi-Shern, 2019-11-28:
@slacmshankar Hi
Some PVs are refreshed normally, but the archive status is always Appliance assigned. Restarting IOC and archive appliance are useless. What is the problem?