archiver-appliance/epicsarchiverap #96: Can we retireve PV disconnect and connect times?

aawdls, 2020-02-14:
With the Channel Archiver, in the ArchiveViewer we used to see little dots indicating times when the PV connection was lost & regained. At Diamond, we'd like to be able to display the equivalent information from our Archiver Appliance in the CSS Data Browser.

I can see that the disconnect / connect times are stored in the protobuf files, e.g..

```
secondsintoyear: 26998393
nano: 380225166
fieldvalues {
  name: "cnxlostepsecs"
  val: "0"
}
fieldvalues {
  name: "cnxregainedepsecs"
  val: "1510227193"
}
fieldvalues {
  name: "startup"
  val: "true"
}
```

I can also get the timestamp of the _last_ disconnect/reconnect events, and a count, from the mgmt  API using `getPVDetails`.

The question is, is there a way to get the full list of connect / disconnect timestamps (through retrieval, or by some other method), in order to display them in the Data Browser?

Thanks!

cc @willrogers

slacmshankar, 2020-02-14:
>> is there a way to get the full list of connect / disconnect timestamps
Yes, you should get this info as part of the JSON/Raw response as an extra field in the "fields" attribute.
"fields": { "cnxlostepsecs": "1580803214",...."cnxregainedepsecs": "1580803228"} 

There is no call yet to retrieve just this information; should be easy to do as a post processor. Is this what you are looking for?

aawdls, 2020-02-17:
Hi Murali, 

Thanks - I see. I have reproduced this for an example PV using the PB raw protocol in python, and can see the extra fields. Probably we'd just want to extract this information from the response we are already getting in CSS, to save making a second API request.

However, for getting this information alone, out of curiosity: when you refer to post-processor, do you mean using this capability: https://slacmshankar.github.io/epicsarchiver_docs/userguide.html#post_processing ? I can't spot on that page how to access the extra fields - is there a more detailed reference for the post processing I can look at?

Many thanks for your help.

Andy

slacmshankar, 2020-02-19:
See the JavaDoc for more info on PostProcessors - http://slacmshankar.github.io/epicsarchiver_docs/api/org/epics/archiverappliance/retrieval/postprocessors/PostProcessor.html

We implement this interface and then register in the PostProcessors class; we expected that folks would implement a bunch of these for their specific purposes.

Please let me know if you need more info.

closed 2020-02-19 by slacmshankar