archiver-appliance/epicsarchiverap #99: Display garbled problem

HongYi-Shern, 2020-04-14:
Hi,
The year and date on the webpage are displayed normally, and the month is displayed as? ?,I don’t know what the problem is. It used to be normal. I recently restarted the host.

slacmshankar, 2020-04-14:
Which web page? A screenshot might help; I can point you to the code that displays the offending element and we can start debugging from there.

HongYi-Shern, 2020-04-14:
[image]
please check the screenshoot!

slacmshankar, 2020-04-14:
This is the code that does this rendering. For some reason, the month is coming out as ??. Would you have a better notion of why? Maybe something to do with your LOCALE..

https://github.com/slacmshankar/epicsarchiverap/blob/master/src/main/org/epics/archiverappliance/common/TimeUtils.java#L156