archiver-appliance/epicsarchiverap #103: PV shows that the connection is normal, but in fact it is not connected

HongYi-Shern, 2020-06-19:
Some PV connection status shows that the connection is normal, but the Last event column shows never.In fact, the PV release is normal, but the archiver appliance is not connected.Please check the picture below for details.

[image]

RuizheIsaf, 2021-11-30:
hi, @HongYi-Shern 
do you resolve your question? i met the same question...

slacmshankar, 2024-11-15:
See if pause/resume reconnects and gets data. If it still doesn't, pause and then check the PVDetails and see if there are any _Channels still hanging around_ Finally, see if a restart fixes the issue.