archiver-appliance/epicsarchiverap #129: http 404 error when i click the button "quick chart"

RuizheIsaf, 2021-11-19:
hi  everyone,
As the figure show, "404 not found" occur when i clicked the button "Quick chart" in order to show the history data of a PV, i think perhaps that i don't have the software "archViewer"? or some problems of Tomcat? I'm not sure if this is a problem with the environment configuration, but other features work well, could you help me to check the problem.
looking forward to your reply and thanks a lot
Ruizhe
[image]
[image]
[image]

aawdls, 2021-11-19:
Have you looked in the log files (usually going to `/var/log/tomcat/`)? I would look in the retrieval log first. Can you retrieve data by other means, e.g. JSON and PB via the API?

The quick chart is provided by the svg_viewer git submodule. Is it possible that this was not included in your installation?

jeonghanlee, 2021-11-20:
Please check your XML file especially **data_retrieval_url** to compare to the following guide section _Create an appliances.xml_ http://slacmshankar.github.io/epicsarchiver_docs/installguide.html

Your front-end to end-users should be one of two instances or one of an independent one. 

HTH
@jeonghanlee

RuizheIsaf, 2021-11-21:
@aawdls @jeonghanlee 
thanks for your help~~
      I have found the reason, because the loss of package "svg_viewer". I git clone the epicsarchiver-src.zip directly but the directory "svg_viewer" is empty, then i check the "MakeFile" written by @jeonghanlee and find that perhaps "svg_viewer" need to be downloaded by using "git submodule update "(in fact, i don't know why we need to this...)
     what's more, i really don't understand how epics AA distribute PV to each appliance of a cluster(which has 3 ~ 5 appliances), according to RAM or CPU capacity...? If an appliance crash, the PVs which are archived in this appliance will be moved to other appliances automatically?
     Thanks again for your time and help!
Ruizhe

jeonghanlee, 2021-11-21:
Not for this repository, but in a different repository, 

`make init` calls `git submodule update` within the rule. It looks like there is something wrong in somewhere.

https://github.com/jeonghanlee/epicsarchiverap-env/blob/ade5e99a6b4229278b03634cca6d1df6f0b38963/configure/RULES_SRC#L29

HTH,
Han

jeonghanlee, 2022-05-04:
@slacmshankar I think, he found the solution. We can close this.

RuizheIsaf, 2022-05-05:
@slacmshankar @jeonghanlee 
yes, i have found the solution, thank you for your helps.

closed 2022-05-05 by RuizheIsaf