archiver-appliance/epicsarchiverap #130: PV may not exist but i can watch the data by using "caget"

RuizheIsaf, 2021-11-24:
hi everone,
There are some PV that i cannot archive in EPICS AA, some details are shown in the figure, i check the network works well, and i can watch the data by using "caget"...
could you give me some help to find what happen?
thanks a lot
Ruizhe
[image]

aawdls, 2021-11-24:
- Can you caget the PV from the archiver appliance machine? In the screenshot it looks like you may be on a different machine when doing e.g. cainfo.
- Are the IOC and the archiver appliance on the same network?
- Are the EPICS_CA_ADDR_LIST and similar variables set correctly on the archiver host?
- Can the archiver connect to other PVs on the same host and/or network?
- Is the timestamp of the PV in the IOC correct?

RuizheIsaf, 2021-11-24:
@aawdls thanks for your reply
1. Yes, i can caget these PV(BL:SP:Ang and BL:SP:Line:Y) from machine 192.168.20.128 which is ip address of an appliance of my epics AA cluster.
2. No, these two IOC are in the network 192.168.30.10, and my epics AA cluster has two appliances(192.168.20.128 and 192.168.20.130), but i have set the network of cluster, in 192.168.20.128, i can caget and ping these two PV...
3. EPICS_CA_ADDR_LIST = localhost
    EPICS_CA_AUTO_ADDR_LIST = YES
    EPICS_CA_MAX_ARRAY_BYTES = 16384
    two appliances have the same configuration like that.
4. Yes, actually i have archived hundreds of PV, but these IOC are all in the network 192.168.20.*.  these two PV are in the net work 192.168.30.*
5. Actually i am not good at writing db files, but i think the timestamp of IOC should be right, because they are used for many months, if necessary i will check with my colleague.
thanks a lot 
Ruizhe

closed 2022-02-18 by RuizheIsaf