archiver-appliance/epicsarchiverap #134: Ability to Export other fields EGU, DESC etc in the JSON request

Mezot1, 2022-03-15:
Is this possible to be implemented or is there somewhere in the code base to add it?  When doing  request to the AA, the JSON export only includes the value, alarm severity, timestamp.  It would be great to export some other fields like EGU, DESC etc that are included in records.

aawdls, 2022-03-15:
Hello - you can use the `fetchLatestMetadata` argument to the retrieval request to make it return the values of these extra fields. For some more info search for this parameter in the user guide: https://slacmshankar.github.io/epicsarchiver_docs/userguide.html

Mezot1, 2022-03-15:
Thank you for the response, I have seen that documentation but I think that only applies to the getData endpoint and not the getDataAtTime. Sorry I forgot to include that in the original post.  I have tried it with the getDataAtTime but no luck I have tried to tack on &fetchLatestMetaData=true (hopefully this is correct way to do it) to the http request.

aawdls, 2022-03-16:
The capitalisation is important, i.e. `fetchLatestMetadata=true`.

I have only used this argument with `getData`. I have not used `getDataAtTime` before. I just tried it, but I could not make it work with `fetchLatestMetadata`. So it seems this argument is not supported with `getDataAtTime`.

slacmshankar, 2024-06-19:
Some progress on this. The next release of the appliance should have some variation on this

```
{
  "A:PV:NAME": {
    "severity": 0,
    "val": 5.398787021636963,
    "nanos": 487564097,
    "meta": {
      "EGU": "KV",
      "HIHI": "0.0",
      "HIGH": "0.0",
      "LOW": "0.0",
      "LOLO": "0.0",
      "PREC": "2",
      "LOPR": "0.0",
      "HOPR": "0.0",
      "DBRType": "DBR_SCALAR_DOUBLE",
      "DESC": "Pump Voltage"
    },
    "secs": 1718816399,
    "status": 0
  }
}

```

closed 2024-09-25 by slacmshankar