archiver-appliance/epicsarchiverap #139: Elog configuration

katysaintin, 2022-07-18:
I saw that now Appliance also support for posting to an elog.
Could you help me to test this new feature ? Where can configured the URL of an Elog server ?

Should we develop a web page  /postToElog URL ?

Otherwise, is Olog works instead of Elog ?

Thank you for your help.

Katy

slacmshankar, 2022-10-04:
This requires some server side support; that is, a small stub that receives a HTTP POST request with "comment", "link" and "snapshot" parameters and does the needful. Our implementations here post to the Physics eLog; the stub is a small Python app that receives the multipart mime POST request and then dumps it into XML files.

```
// We support posting to an elog; by default we post to a "postToElog" endpoint relative to the viewer.
// The endpoint gets a multipart mime document with "comment", "link" and "snapshot" elements.
viewerVars.postToElogURL = "../../../postToElog";
```

closed 2022-10-04 by slacmshankar

nayanbera, 2024-06-24:
Would it be possible to share the python App implementation to post to the Elog?