archiver-appliance/epicsarchiverap #141: Python 2.7 is past end of support

prjemian, 2022-09-12:
Support for Python 2.7 has [ended officially](https://www.python.org/doc/sunset-python-2/).  The [Python release schedule ](https://www.python.org/downloads/) shows the oldest supported release is 3.7.

It's time to update the use of Python to the current release schedule.

Support for Python versions here varies between Py2.7 and Py3.  Amongst the syntax changes, the `print` statement in Py2.7 has changed to a `print()` function in Py3.  This use can fingerprint Python files in need of upgrade to Py3.  Another such fingerprint is in the first line where `/usr/bin/env` is called:

```bash
USER@HOST$ grep /usr/bin/env $(locate slacmshankar/epicsarchiverap | grep ".py\$")
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/abortNeverConnectedPVs.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/addMysqlConnPool.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/addPostProcessingOperator.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/archiveFromDB.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/archivedPVsNotInList.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/changeArchiveStore.py:#!/usr/bin/env python3
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/checkConnectedPVs.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/checkForEngineActivity.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/checkTypeChangedPVs.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/consolidateArchivedData.py:#!/usr/bin/env python3
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/consolidatePausedPVs.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/deletePVList.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/deployMultipleTomcats.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/getPVList.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/listTypeChanges.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/pausePVList.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/pingCurrentlyDisconnectedPVs.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/printCurrentlyDisconnectedPVs.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/removeMetaFields.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/renamePVList.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/resumePVList.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/resumePausedPVsMatchingPattern.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/stopArchivingCurrentlyDisconnectedPVs.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/storageSizeCheck.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/unarchivedPVs.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/debug/sumUpETLTimesFromPVDetails.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/debug/typeInfoFromBackup.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/docs/samples/pythonPBRaw/printDBRDouble.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/src/test/org/epics/archiverappliance/retrieval/postprocessor/data/deadband/deadband_updates.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/src/test/pythontests/CompareBoth.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/src/test/pythontests/SimplePlot.py:#!/usr/bin/env python
GITHUB_REPO/slacmshankar/epicsarchiverap/src/test/pythontests/StartEndTime.py:#!/usr/bin/env python
```

(Two of these 32 files call for python3.)

prjemian, 2022-09-12:
And here (where this was first noticed):  https://github.com/slacmshankar/epicsarchiverap/blob/d3dd9cc144004ab84b49b6d7fcd5c575d67404d8/docs/samples/single_machine_install.sh#L333

slacmshankar, 2022-10-05:
I agree; fairly large task but will try to wrap up by the next release.

minijackson, 2022-10-06:
@slacmshankar you might be able to use `2to3` to automate the conversion, it has worked quite well for me:

```bash
2to3 -w -f all -f buffer -f idioms -f set_literal --no-diffs .
```

Doing this on the codebase works (although I haven't the Python code), the only issue I found is that it double wraps `print` arguments in parentheses in code that is already Python3.