archiver-appliance/epicsarchiverap #180: Retrieval of deleted data

GDH-ISIS, 2023-09-18:
Using the BPL api, we have worked out that PVs that have changed type can be paused, consolidated, renamed to a different name, deleted (keeping the data) and finally ask for the PV to be archived once more using the original name. The new archived data then looks to have the correct name and type (as recommended https://slacmshankar.github.io/epicsarchiver_docs/admin.html). However, may I ask whether it is possible to retrieve the old data that is now under a different name using the BPL api. The files look to be present in the storage area, but not in any of the PV lists. Any advice would be appreciated.

jacomago, 2023-09-22:
Yes, there are two solutions:

Feature org.epics.archiverappliance.retrieval.SearchStoreForRetiredPvs in the archappl.properties file for searching for retired PVs. PVs that are not longer being archived.

Some other options in https://slacmshankar.github.io/epicsarchiver_docs/api/mgmt_scriptables.html include the

- /appendAndAliasPV endpoint
- /addAlias endpoint

For changing names of PVs that you are archiving.

GDH-ISIS, 2023-10-09:
My apologies for not having read this sooner. This looks like just what I am looking for. Much appreciated.

closed 2023-10-09 by GDH-ISIS