archiver-appliance/epicsarchiverap #184: Is there a way to speed up the initial monitoring period?

jembishop, 2023-10-18:
Hi I am using the EPICS archiver appliance as part of a end to end testing setup.
Is there a way to speed up the intial monitoring period? I just have a few test PVS I want to insert and 
I would prefer if they were tracked instantly.

jembishop, 2023-10-18:
Thanks for getting back to me. I was refering to the around 5 minute delay between clicking the archive button, and the PV actually being started archived. It looks like the policy.py file would probably be the best way to go here, I will try that. Thanks again

jembishop, 2023-10-18:
Would you mind helping me out with my policies.py file? This is what I think I need, its not working however 
it gives me ```An error occured on the server while requesting these PVs to be archived -- error --``` on the frontend and I can't see anything in the logs.
```
def getPolicyList():
	pvPoliciesDict = {}
	pvPoliciesDict['Default'] = 'The default policy'
	return pvPoliciesDict

# Define a list of fields that will be archived as part of every PV.
# The data for these fields will included in the stream for the PV.
# We also make an assumption that the data type for these fields is the same as that of the .VAL field

# We use the environment variables ARCHAPPL_SHORT_TERM_FOLDER and ARCHAPPL_MEDIUM_TERM_FOLDER to determine the location of the STS and the MTS in the appliance
shorttermstore_plugin_url = 'pb://localhost?name=STS&rootFolder=${ARCHAPPL_SHORT_TERM_FOLDER}&partitionGranularity=PARTITION_HOUR&consolidateOnShutdown=true'
mediumtermstore_plugin_url = 'pb://localhost?name=MTS&rootFolder=${ARCHAPPL_MEDIUM_TERM_FOLDER}&partitionGranularity=PARTITION_DAY&hold=2&gather=1'
longtermstore_plugin_url = 'pb://localhost?name=LTS&rootFolder=${ARCHAPPL_LONG_TERM_FOLDER}&partitionGranularity=PARTITION_YEAR'
#longtermstore_plugin_url = 'blackhole://localhost'

def determinePolicy(pvInfoDict):
    pvPolicyDict = {}
    pvPolicyDict['samplingPeriod'] = 1.0
    pvPolicyDict['samplingMethod'] = 'MONITOR'
    pvPolicyDict['skipCapacityPlanning'] = True
    pvPolicyDict['dataStores'] = [
        shorttermstore_plugin_url, 
        mediumtermstore_plugin_url, 
        longtermstore_plugin_url
        ]
    pvPolicyDict['policyName'] = 'Default'
    pvPolicyDict["archiveFields"]=[]

    return pvPolicyDict
```

jacomago, 2023-10-19:
Maybe you need to define the function 

def getFieldsArchivedAsPartOfStream():
    return []

There is a separate log file for errors from the policies.py file. Something like /var/log/tomcat/arch-policy.log that might give you useful error output.

I would also set the "appliance" field, as it should reduce some of the logic needed before archiving.

closed 2024-02-02 by jembishop