archiver-appliance/epicsarchiverap #188: Changing policy on existing (archiving) PVs

justinabraham, 2023-11-06:
Hi,

I've deployed an instance of the archiver using [Jeong Han Lee's](https://github.com/jeonghanlee/epicsarchiverap-env) EPICS Archiver appliance configuration environment and it's working well. Thank you for the great tool.

I edited the default policy file before deployment and changed the default policy settings, an example is:

```
elif '5sec-3year' in userPolicyOverride:
        pvPolicyDict['samplingPeriod'] = 5.0
        pvPolicyDict['samplingMethod'] = 'SCAN'
        pvPolicyDict['dataStores'] = [
            STS_BASE_URL,
            MTS_BASE_URL,
            BLACKHOLE_URL
        ]
```
I'm currently archiving a number of PVs with this policy applied. I subsequently realised I need to change the policy settings, like samplingPeriod or modify dataStores etc.

I changed the policy file and restarted the archiver and I can archive new PVs and the policy is applied correctly (updated parameters). However, the existing PVs (before policy settings were changed) are still archived with the previous policy parameters. I've tried to pause a PV, restart archiver etc. but no luck.

How do I change a policy such that those changes are reflected in currently archived PVs?

I would appreciate any inputs, thanks.

jacomago, 2023-11-06:
Hi,

Currently there is no way to reanalyse the PVs with the policy file and reset the configuration of each PV.

I think you have two main options:

1) Manually edit the configuration of each PV via the web interface or via the api endpoints:

- /modifyStoreURLForPV  https://slacmshankar.github.io/epicsarchiver_docs/api/org/epics/archiverappliance/mgmt/bpl/ModifyStoreURLForPV.html 
- /changeArchivalParameters https://slacmshankar.github.io/epicsarchiver_docs/api/org/epics/archiverappliance/mgmt/bpl/ChangeArchivalParamsAction.html 

2) Pause, then delete (without deleting data) all the PVs, then readd them to be archived. This will keep the data in the original locations, and update the configuration of the PVs from your new policy file.

Hope that helps.

justinabraham, 2023-11-06:
Hi,

Thanks for the response. I did try the second approach. I pause a PV, delete, don't tick checkbox to delete data then re-add to archiver with updated policy.

It's archiving as per updated policy but I can't access old data via quick chart. Only data from when it was re-added.

Could a name change to dataStore cause such an issue?

jacomago, 2023-11-06:
Yes, updating the policy won't move the data between datastores. The data should still be in the original data stores you specified.

justinabraham, 2023-11-07:
Thanks. I would like to try option 2 - delete pvs.

I see there's a http://mgmt_url/bpl/deletePV endpoint as well. Can I just post to that endpoint to delete a PV? I'd like to run it in a loop to delete multiple pvs.

How does one pass pv name to endpoint, json, url parameter?

closed 2023-11-09 by justinabraham