archiver-appliance/epicsarchiverap #197: A PV and any data related to it can be deleted without any verification

lvhuihui1988, 2023-12-20:
Anyone can delete a PV and its historical archiving data from the website without any password or other permission authentication. If someone dose it by mistake, all the data is lost. Is there any protective measure, for example, adding a password verification? Thanks a lot.

jacomago, 2024-01-08:
Yes, right now there is no authentication or authorization mechanism. There are some options however:

- Only expose the retreival interface to your users, either with proxying the retreival or only allowing uses access via some other interface such as phoebus.
- Setting up a proxy that does authentication, such as apache.
- Setting up the host tomcat service with authentication, see https://serverfault.com/questions/192000/quick-way-to-password-protect-tomcat

slacmshankar, 2024-03-27:
Agree with @jacomago here. We've always intended authnz to be external to the appliance.

closed 2024-03-27 by slacmshankar