archiver-appliance/epicsarchiverap #218: Recover the ant building system

jeonghanlee, 2024-03-17:
@slacmshankar and @jacomago 

Could you recover the ant building system was removed at https://github.com/archiver-appliance/epicsarchiverap/pull/176/commits/44fba7b9bf04d9d742b81de99f2ed91f0e00b5f5 ?

I had the very naive thought on Gradle could be a yet another version of ant or maven. And today, I spent a couple of hours, then I realized that I was completely wrong.

The main issue I have is that I don't have enough time to understand this building system, since this is not for only java. I have enough works to understand `Makefile` and `maven`. And I don't want to understand the Gradle completely, since it allows to do something which I don't want to do during compile Java applications. 

Gradle is a nice tool, I agree, but it is not for everyone who is the community. 

Please just return the ant building system as an option for the old generation.

jacomago, 2024-03-18:
Hi @jeonghanlee .

I'm sorry for creating your confusion. @shroffk has also brought up the idea of trying to reduce the number of build systems the java side of epics use. I am welcome for anyone to convert the build system to maven and would support that. Or if people would prefer to move to gradle, I have no particular preference. 

From the other posts you created, it looks like the problem you are facing is when inserting site templates into the site specific folder. I think I can fix this easily with an ignore folder in the build script. I'll work on this today. Please send me any other errors you encounter and I'll try fix them. 

In the future, my aim is to not have non maintainers compile or build the program. I think this would produce a more stable program where there is more configuration than customization. Unfortunately pulling apart the site specific build steps takes time, which I haven't prioritised. Since here at ESS we do the html templating changes after the build. I will try prioritize it  now.

Apologies again for the difficult migration. I have updated the documentation inside of the docs folder of the repository. I haven't updated the website as I only found out that it wasn't linked recently. That's hopefully a project we'll work on soon.

jacomago, 2024-03-18:
@jeonghanlee Can you try branch https://github.com/jacomago/epicsarchiverap/tree/site-spefici-gradle ?

jacomago, 2024-03-19:
@jeonghanlee The ant builds were broken so we swapped to using only the gradle one. Have you tried the branch https://github.com/jacomago/epicsarchiverap/tree/github-test-site-specific which I think could fix your problem?

jeonghanlee, 2024-03-20:
- The ant build works at 8b4f22e (2024-01-23), after this commit, you remove the ant build. Please revert your commit.
- I am not talking to "the site specific build"

jeonghanlee, 2024-03-20:
@slacmshankar do you not work on here anymore? Now did ESS take over the repository completely?

jacomago, 2024-03-20:
@jeonghanlee There was an issue where the ant build was creating wars which wouldn't then run. So yes the ant build system worked, but it did not create usable war files. 

I've emailed you and @slacmshankar to discuss this ticket.

jeonghanlee, 2024-03-20:
@jacomago Okay, I accepted this unacceptable situation. No document, no fix issues, and merged the broken into the masters, and so on. 

Please fix a gradle build system not gradle wrapper.

closed 2024-03-22 by jeonghanlee

slacmshankar, 2024-03-27:
Apologies; I was on vacation. I'm not sure re-adding the ant build system is worth it. One of the main issues I had with that was the complexity of upgrading the dependent jar files. I'm spending a fair amount of time catching up on technical debt and that main reason is because of the Ant build system I think. And I'd like to move to something the rest of the collaboration is using. 

We'll update the documentation during the summer codeathon and hopefully get some of the build specific features back in. In the meantime, plz reach out to Sky et al if you need help with specific build issues.

shroffk, 2024-03-27:
> I'm not sure re-adding the ant build system is worth it. One of the main issues I had with that was the complexity of upgrading the dependent jar files.

+1

I am glad to hear your take murali.

We can definitely work in the documentation and input from Han can be very valuable to ensure the documentation is useful.

I also think we should explore the possibility of moving to maven... for the simple reason that we can then standardize the build for all the EPICS middle layer services and tools