archiver-appliance/epicsarchiverap #284: default build doesn't include all necessary config files

jacomago, 2024-11-01:
See https://epics.anl.gov/tech-talk/2024/msg01220.php 

Maybe should add to CI a way to check the singleMachineInstall.sh script works... or tell people to use the docker image? dunno

jacomago, 2024-11-04:
I think the main problem is the archappl.properties file not being in the default build. Best solution is probably to move towards maven style folders for resources, but that would be a pain in breaking every fork. Better to use easy solution of putting a copy of archappl.properites file in the classpaht of the default build.

slacmshankar, 2024-11-04:
For now, maybe document in detail the additional steps required. We'll fix this for 2.0.6 This and maybe the Matlab response bug warrant a new dot release maybe.

slacmshankar, 2024-11-04:
_Better to use easy solution of putting a copy of archappl.properites file in the classpaht of the default build._

I think this is the best. The "default" build should be something that the quickstart.sh and singleMachineInstall.sh should work against.

slacmshankar, 2024-11-04:
These may be the additional steps

1. Copy a log4j2.xml to apache-tomcat-9.0.xx/lib/log4j2.xml
2. Copy a archappl.properties and policies.py to mgmt/webapps/mgmt/WEB-INF/classes
3. Remove the "-XX:MaxPermSize=128M -XX:+UseG1GC" from sampleStartup.sh; newer JDK's may not start with these arguments

Maybe 1 can be solved by adding a log4j2.xml to sitespecific/default/classpathfiles

slacmshankar, 2024-11-15:
Checked quickstart and it seemed to work for me...

jacomago, 2025-05-09:
Fixed by #285

closed 2025-05-09 by jacomago