archiver-appliance/epicsarchiverap #324: "optimized" produces non-monotonic result

mdavidsaver, 2025-01-29:
I am seeing situations where the "optimized" operation yields results with duplicate, out of order, samples.

See attached queries of a -1 day span for:

- [`ext:1:T-I`](https://github.com/user-attachments/files/18590954/raw.txt)
- [`caplotbinning_5760(ext:1:T-I)`](https://github.com/user-attachments/files/18590958/caplotbinning.txt)
- [`optimized_5760(ext:1:T-I) `](https://github.com/user-attachments/files/18590959/optimized.txt)

This last file ends with:

```
01-29 06:21:18.816735 optimized_5760(ext:1:T-I) 7.9
01-28 08:01:34.720683 optimized_5760(ext:1:T-I) 3.0
01-29 07:01:05.906998 optimized_5760(ext:1:T-I) 8.2
01-29 07:28:59.963163 optimized_5760(ext:1:T-I) 8.4
01-28 08:01:34.720683 optimized_5760(ext:1:T-I) 3.0
01-28 08:01:34.720683 optimized_5760(ext:1:T-I) 3.0
```

showing the `01-28 08:01:34.720683` sample repeated three times.  The raw and caplotbinning results are in order, without duplicates.

mdavidsaver, 2025-01-29:
I am testing with AA 2.0.5 (git tag).

GDH-ISIS, 2025-01-29:
Unsure if this helps, but I believe I observed something that sounds like the same issue when using the Phoebus DataBrowser. Latest EAA commit/build on Docker. Optimized data appeared to have horizontal discontinuities. Raw data came through fine. I wondered whether a new setting was required and built to a previous commit instead which worked (commit ea4da03c0e64e6ce744179fa609aefb92b8d9fe1)

mdavidsaver, 2025-01-29:
> ... using the Phoebus DataBrowser ...

This is is the context in which I first noticed this behavior.  Here is one example:

[image]

GDH-ISIS, 2025-01-29:
This is exactly my observation also.

jacomago, 2025-01-30:
That's certainly a fun one. Do you have the raw protobuf file that the data is coming from?

mdavidsaver, 2025-01-30:
Will the raw query result be sufficient?  [raw.pb.txt](https://github.com/user-attachments/files/18604227/raw.pb.txt) 
[optimized.pb.txt](https://github.com/user-attachments/files/18604231/optimized.pb.txt)

```
curl 'http://appl:17665/retrieval/data/getData.raw?pv=ext:1:T-I&from=2025-01-29T14:47:16.962633Z&to=2025-01-30T14:47:16.962633Z' > /tmp/raw.pb.txt

curl 'http://appl:17665/retrieval/data/getData.raw?pv=optimized_5760(ext:1:T-I)&from=2025-01-29T14:47:16.962633Z&to=2025-01-30T14:47:16.962633Z' > /tmp/optimized.pb.txt
```

slacmshankar, 2025-01-31:
I can reproduce this on any old PV I think

slacmshankar, 2025-01-31:
I have a PV mshankar:arch:static1 that I have to manually change (SCAN is not defined). I update this a couple of times such that the raw .txt returns 
```
Sep/27/2024 10:41:00 PDT        1.0     0       0       151420375
Jan/31/2025 15:54:38 PST        1.0     0       0       32153808
Jan/31/2025 15:54:42 PST        2.0     0       0       734133765

```
With the same data, 
`getData.txt?pv=optimized_10(mshankar:arch:static1)`
returns
```
Sep/27/2024 10:41:00 PDT        1.0     0       0       151420375
Jan/31/2025 15:54:38 PST        1.0     0       0       32153808
Jan/31/2025 15:54:42 PST        2.0     0       0       734133765
Sep/27/2024 10:41:00 PDT        1.0     0       0       151420375
Sep/27/2024 10:41:00 PDT        1.0     0       0       151420375
```
Does this reproduce this bug?

mdavidsaver, 2025-02-01:
> Does this reproduce this bug?

I think so.

slacmshankar, 2025-02-05:
Most likely this is from 
https://github.com/archiver-appliance/epicsarchiverap/blob/92c7e775b27008fe31d669fc72152237bf6be10e/src/main/org/epics/archiverappliance/retrieval/postprocessors/SummaryStatsPostProcessor.java#L233

where we are trying to add the last event. I don't exactly know what this postprocessor is supposed to do but there's a broken boundary condition here. Will continue to dig into it

mdavidsaver, 2025-03-02:
> Most likely this is from

git-blame points to 8b4f22efe435d08c7f1bdf04f4b189ef49018f70 by @abrahamwolk ?

GDH-ISIS, 2025-03-06:
@abrahamwolk - via phoebus, I believe your fix looks to have worked. My dataset is small, single archiver, running in docker envionment. That said, if I build to the latest commit, I have odd horizontal lines with optimized data. Building to your fix does look to solve the issue. 

[image]

mdavidsaver, 2025-03-07:
> Can you replicate this kind of output and also test it with the fix I proposed?

The case at cited about now looks reasonable.  (I confirmed again that it did not prior to updating AA)

[image]

> Do you run several concurrent instances of the archiver?

No.

> I don't have an explanation for why you also get duplicates ...

Ok.  I will continue to look for oddities.

GDH-ISIS, 2025-03-07:
Happy to test. I believe my testing would probably only have tested the short term storage. @abrahamwolk - you raise a good point. What I can mention is that on our live system, we have build to an old commit - ea4da03c0e64e6ce744179fa609aefb92b8d9fe1 (June 27th 2023! - again docker). This does not appear to have any problems and does cover short, medium and long term storage and the same Phoebus client.

slacmshankar, 2025-03-16:
abrahamwolk, please feel free to close when appropriate

mdavidsaver, 2025-04-07:
@abrahamwolk who is your question directed to?

jacomago, 2025-04-09:
I've made a release (2.0.8) including https://github.com/archiver-appliance/epicsarchiverap/pull/333 for until we come up with a better solution.

mdavidsaver, 2025-08-03:
I do not think this issue has been completely addressed.  When looking at 3 year time range, I have a case where it looks like two years are being folded onto each other.  My archiver instance has the default yearly LTS, so I think that `PayloadInfo` at the start of each year is being omitted.

`optimized_5760()`

[image]

Raw

[image]

jacomago, 2025-08-04:
Do you only see this on a span of more than a year? Or does it also happen across a year boundary?

jacomago, 2025-08-05:
Great thanks, that should help creating a test case.

jacomago, 2025-08-05:
That case also is across 1 January, in particular with empty bin over that time period. Do you have any examples without that?