archiver-appliance/epicsarchiverap #331: Access Control Mechanism for UI Pages

XavSPM, 2025-02-12:
Hello,

I would like to know if there is a security mechanism to control access to the pages `mgmt/ui/index.html` and `retrival/ui/viewer/archViewer.html`, such as a login/password system.

Thank you in advance for your response.

slacmshankar, 2025-02-14:
We do not have built in integrations for authn/authz. Since these are web services, we can use web server authentication mechanisms like Apache's mod_auth modules - https://httpd.apache.org/docs/2.4/howto/auth.html.

closed 2025-02-14 by slacmshankar

XavSPM, 2025-02-26:
@slacmshankar  Thank you for this information.

We will test this and get back to you with the results. If the results are conclusive, it might be interesting to add this information to the documentation. What do you think?