archiver-appliance/epicsarchiverap #362: Remove the dependency on Selenium for the unit tests

slacmshankar, 2025-06-06:
While this does test the UI, it has all these X dependencies that make testing in modern environments difficult. For most tests, we can simply use the BPL or `putPVTypeInfo`. Also using `awaitability` ( thanks @jacomago for this tip) will speed up most tests significantly. 

This is a catch all issue; that is, there'll be multiple PR's for this issue spread over some time. 

Any thoughts?

jacomago, 2025-06-11:
I totally agree!

slacmshankar, 2025-06-14:
These are the tests that need to be refactored

- org/epics/archiverappliance/mgmt/InactiveClusterMemberArchivePVTest.java
- org/epics/archiverappliance/mgmt/ChangeArchivalParamsTest.java
- org/epics/archiverappliance/mgmt/ModifyMetaFieldsTest.java
- org/epics/archiverappliance/mgmt/VALNoVALTest.java
- org/epics/archiverappliance/mgmt/pauseresume/ResumePVAfterRestartTest.java
- org/epics/archiverappliance/mgmt/pauseresume/DeletePVAfterRestartTest.java
- org/epics/archiverappliance/mgmt/pauseresume/DeletePVTest.java
- org/epics/archiverappliance/mgmt/pauseresume/DeleteMultiplePVTest.java
- org/epics/archiverappliance/mgmt/ScanSamplingMethodTest.java
- org/epics/archiverappliance/mgmt/ArchivePVTest.java
- org/epics/archiverappliance/mgmt/AddRemoveAliasTest.java
- org/epics/archiverappliance/mgmt/ArchiveFieldsWorkflowTest.java
- org/epics/archiverappliance/mgmt/appxml/FQDNApplianceXMLTest.java
- org/epics/archiverappliance/mgmt/appxml/LocalhostApplianceXMLTest.java
- org/epics/archiverappliance/mgmt/appxml/IPApplianceXMLTest.java
- org/epics/archiverappliance/mgmt/ArchiveAliasedPVTest.java
- org/epics/archiverappliance/mgmt/RenamePVBPLTest.java
- org/epics/archiverappliance/mgmt/ArchiveFieldsNotInStreamTest.java
- org/epics/archiverappliance/mgmt/ArchiveWithSamplingPeriodPVTest.java
- org/epics/archiverappliance/mgmt/ArchivePVClusterTest.java
- org/epics/archiverappliance/retrieval/cluster/ClusterAliasTest.java
- org/epics/archiverappliance/retrieval/cluster/ClusterAliasSpanApplianceTest.java
- org/epics/archiverappliance/retrieval/postprocessor/VariousPostProcessorTest.java
- org/epics/archiverappliance/retrieval/extrafields/EGUChangeTest.java
- org/epics/archiverappliance/retrieval/client/Mean600Test.java
- org/epics/archiverappliance/retrieval/client/PostProcessorWithPBErrorDailyTest.java
- org/epics/archiverappliance/retrieval/client/PostProcessorWithPBErrorTest.java
- org/epics/archiverappliance/engine/test/CnxLostTest.java
- org/epics/archiverappliance/reshard/BasicReshardingTest.java

closed 2025-06-29 by slacmshankar