archiver-appliance/epicsarchiverap #401: How to enforce new policies.py to archived PVs

caraxlr, 2025-08-04:
Hi,

We've updated policies.py to change the sampling period and method for PVs. However, after relaunching the AA with the new policies.py, the previously archived PVs still retain the sampling settings defined by the old version.

I was wondering if there is a way to enforce the updated policies.py settings on already archived PVs, or is the only option to manually update their sampling parameters or rearchive them?

Thank you,
Lingran

jacomago, 2025-08-05:
I think rearchiving is the easiest way.

Steps:

1. get name and appliance of each pv
2. delete pv from archiver (without deleting data)
3. submit pv specifiying the appliance (this way the data will stay linked)

 But thats a feature idea I've thought about a few times. And would be very useful for the channelfinder integration.

caraxlr, 2025-08-05:
Yes, I think it’s a natural thought to recalculate the sampling period/method based on the updated policies.py. It would be very helpful to have the option to re-archive PVs according to the new policies.py.

BTW, what’s the current state of the ChannelFinder integration with AA? Is there any relevant documentation available?

Thank you!